open-power/snap #729: VIV2018 + xcelium: snap_maint hangs in unlock_action

joergkayser, 2018-04-19:
export SNAP_TRACE=0xFF
hdcl070: jkayser /afs/bb/proj/fpga/framework/jkayser/snap_modelsim/hardware/sim/ies/20180419_175200 $ snap_maint -vvv
[main] Enter
[snap_open] Enter: /dev/cxl/afu0.0m
D hw_snap_card_alloc_dev Enter /dev/cxl/afu0.0m
INFO:Connecting to host 'hdcl070.boeblingen.de.ibm.com' port 16384
D   hw_snap_card_alloc_dev: errinfo_size: 1 VendorID: ffff DeviceID: ffff
D hw_snap_card_alloc_dev Exit 0x16b4010 OK Context: 0 Master: 1
[snap_open] Exit 0x16b4010
[snap_version] Enter
D   hw_card_ioctl CARD_TYPE: 0
D   hw_card_ioctl NVME: 0
D   hw_card_ioctl Get MEM: 1 MB
SNAP on ADKU3 Card, NVME disabled, 1 MB DRAM available.
R   hw_snap_mmio_read64(0x16b4010, 0, 103054c59d0c4f9) 0
SNAP FPGA Release: v1.3.5 Distance: 76 GIT: 0x59d0c4f9
R   hw_snap_mmio_read64(0x16b4010, 8, 201804191226) 0
SNAP FPGA Build (Y/M/D): 2018/04/19 Time (H:M): 12:26
R   hw_snap_mmio_read64(0x16b4010, a0, 8000000000000000) 0
SNAP FPGA CIR Master: 1 My ID: 0
R   hw_snap_mmio_read64(0x16b4010, 80, 139) 0
SNAP FPGA Up Time: 0 sec
[snap_version] Exit
[snap_m_init] Enter
R   hw_snap_mmio_read64(0x16b4010, 20, 0) 0
R   hw_snap_mmio_read64(0x16b4010, 18, 0) 0
[unlock_action] Enter

ibm-genwqe, 2018-04-20:
Accessing SNAP_ACTION_ID_REG (Offset 0x10) is not returning to software:

```
VERBOSE2("[%s] Enter\n", __func__);
reg = snap_read32(handle, addr + SNAP_ACTION_ID_REG);
```

joergkayser, 2018-04-20:
@ThomasFuchs can you please check, why the access to the action is not working ?
Please note, that this is first of a kind testing with Viv2018.1 and Xcelium 17. 10.002
Your setup can be made with 'ibm_snappings_for_snap xcelium'

joergkayser, 2018-04-25:
@ThomasFuchs  any updates on this ?
error repeated in /afs/bb/proj/fpga/framework/jkayser/snap_modelsim/hardware/sim/xcelium/20180425_091253

joergkayser, 2018-04-30:
the action did not exist due to a bug in the path settings. Fixed

closed 2018-04-30 by joergkayser