open-power/snap #731: Vivado 2018.1 + N250SP + hls_sponge: action compilation fails

joergkayser, 2018-04-24:
make model fails with
[CONFIG ACTION HW....] start 14:47:41 Tue Apr 24 2018
grep: hlsSponge_xcku15p-ffva1156-2-i/sponge/syn/vhdl/hls_action_ctrl_reg_s_axi.vhd: No such file or directory
make[5]: *** [check] Error 1
make[4]: *** [hw] Error 1
make[3]: *** [action_hw] Error 2
make[2]: *** [.hw_project_done] Error 2
make[1]: *** [model] Error 2
make: *** [model] Error 1

make -C actions/hls_sponge fails with
...
INFO: [BIND 205-101] Exploring resource sharing.
INFO: [BIND 205-101] Binding ...
make[1]: *** [check] Error 1
make[1]: Leaving directory `/afs/vlsilab.boeblingen.ibm.com/proj/fpga/framework/CI_ws/vol2/SNAP_SIM_matrix/search_intersect_sim/ACT/hls_sponge/CRD/N250SP/SIM/irun/VIV/2018.1/snap/actions/hls_sponge/hw'
make: *** [hw] Error 1

bmesnet, 2018-04-24:
This is directly generated by HLS depending on the name we give in the code. 
Can you try doing a `make` from the actions/hls_sponge/hw directory? This should be giving more information about the failure

bmesnet, 2018-04-24:
hmm your log is showing:

joergkayser, 2018-04-24:
"make -C actions/hls_sponge/hw clean all"     ran fine without problems.
But FPGACHIP should net be a hard definition in the actions subdirectory, as this is dependent on the card used and should com from make snap_config and its output files.
For N250SP it is xcku15p-ffva1156-2-i

joergkayser, 2018-04-27:
There are also some warnings during action compile:
WARNING: [SCHED 204-21] Estimated clock period (4.1062ns) exceeds the target (target clock period: 4ns, clock uncertainty: 0.5ns, effective delay budget: 3.5ns).
WARNING: [SCHED 204-21] The critical path consists of the following:
	'call' operation ('l', action_sponge.cpp:77) to 'test_hexdigit' (3.34 ns)
	'add' operation ('tmp_18', action_sponge.cpp:80) (0.765 ns)
see difference between /afs/bb/proj/fpga/framework/debug -> e sponge_201[78]/snap/hardware/logs/action_make.log

joergkayser, 2018-06-07:
problem fixed

closed 2018-06-07 by joergkayser