open-power/snap #735: Set FPGACHIP for standalone "make" in action/hls_xxxx/hw

luyong6, 2018-05-09:
Hi Sven, Frank, 

There is a small problem. When I select a card of S121B in `make snap_config`
For the first time, `make model` is OK. 

But when I keep modifying the HLS code, I may need many iterations to compile the HLS CPP code. 
Compared to "make model" which also creates IPs,  "make" under `actions/hls_xxxx/hw` directory takes less time. 

In `make model` process, it correctly creates run_hls_script.tcl by setting --  
```
set_part xcku115-flva1517-2-e
```

In my standalone "make" under `actions/hls_xxxx/hw`
There is a small checking for MMIO area, which is done by `actions/hls.mk`
This script will read $FPGACHIP. And now I don't have $FPGACHIP set. 

So there is such an error: 
```
Checking for critical warnings during HLS synthesis ... OK
Checking for reserved MMIO area during HLS synthesis ... grep: hlsUpperCase_xcku060-ffva1156-2-e/helloworld/syn/vhdl/hls_action_ctrl_reg_s_axi
.vhd: No such file or directory
```

It is still looking for the default xcku060 folder. But my folder is xcku115 now.

luyong6, 2018-05-09:
Add @bmesnet

bmesnet, 2018-05-09:
Correct. Issue also reported by Joerg in issue #731

bmesnet, 2018-05-09:
@luyong6 You need to manually change the FPGA type in hls.mk so that 
`#FPGACHIP    ?= xcku060-ffva1156-2-e        --- all KU060 chips`
`FPGACHIP    ?= xcvu9p-flgb2104-2l-e-es1    -- VU9P chip`
`PART_NUMBER ?= $(FPGACHIP)`
Do a `make clean` in the hls_xxx/hw directory and a new `make` should generate a `run_hls_script.tcl `with correct `open_project `and  `set_part` statement

But yes, you are correct, getting the FPGA type directly would remove this issue

luyong6, 2018-05-17:
Shall I just comment `FPGACHIP ?= xcvu9p-flgb2104-2l-e-es1` in hls.mk and submit a pull request?

bmesnet, 2018-05-17:
well this works  but may need some tests.
One thing that may not work is when you get out of a `make clean `and if you try to do a `make `from the action_hw directory, but if you do a `make model/sim/image `from the snap directory it works ok

boekholt, 2018-07-09:
@luyong6 @bmesnet 
Is that now working as expected?

bmesnet, 2018-07-09:
yes it works ok. May be just a slight enhancement could be needed when doing a `make clean` at snap level which displays as many times the part type as existing actions.  Didn't had time to see how to correct it

luyong6, 2018-07-09:
> displays as many times the part type as existing actions

This should have already been fixed in current master. 
by commenting off line https://github.com/open-power/snap/blob/master/actions/hls.mk#L25

bmesnet, 2018-07-09:
ok correct. Didn't pay attention it was corrected. So go for it. All is good. Thx

boekholt, 2018-07-09:
Closing as fixed (via #742 and #752).
Thanks

closed 2018-07-09 by boekholt