open-power/snap #765: Model: make model fails and succeeds after retry

fhaverkamp, 2018-07-06:
```
bash-4.1$ make model
[HW PROJECT..........] start 16:05:27 Fri Jul 06 2018
[CONFIG ACTION HW....] start 16:05:27 Fri Jul 06 2018
[CONFIG ACTION HW....] done  16:05:27 Fri Jul 06 2018
=======================================================
== SNAP ENVIRONMENT SETUP                            ==
=======================================================
Path to vivado          is set to: /afs/bb/proj/fpga/xilinx/Vivado/2017.4/bin/vivado
Vivado version          is set to: Vivado v2017.4.1 (64-bit)
=====Checking path to PSL design checkpoint============
PSL_DCP                 is set to: "/afs/bb/proj/fpga/framework/cards/N250S/current/b_route_design.dcp"
=====Simulation setup: Setting up PSLSE version==========
Setting PSLVER                 to: "8"
=====Simulation setup: Checking path to PSLSE==========
PSLSE_ROOT              is set to: "/afs/bb/u/haver/framework/pslse"
=====ACTION ROOT setup=================================
Setting ACTION_ROOT            to: "${SNAP_ROOT}/actions/hdl_example"
=====Timing limit for FPGA image build in ps============
TIMING_LABLIMIT         is set to: "-200"
=======================================================

=====Content of snap_env.sh============================
export ACTION_ROOT=${SNAP_ROOT}/actions/hdl_example
export PSLVER=8
export SNAP_ROOT=${HOME}/framework/snap
export PSL_DCP=/afs/bb/proj/fpga/framework/cards/N250S/current/b_route_design.dcp
export PSLSE_ROOT=${HOME}/framework/pslse
export CDS_LIC_FILE=5280@auslnxlic01.austin.ibm.com
export TIMING_LABLIMIT="-200"
#export DENALI_USED=FALSE
=======================================================

                        ==============================
                        hardware/Makefile called with:
                        ACTION_ROOT     = /afs/bb/u/haver/framework/snap/actions/hdl_example
                        PSL_DCP         = /afs/bb/proj/fpga/framework/cards/N250S/current/b_route_design.dcp
                        FPGACARD        = N250S
                        FPGACHIP        = xcku060-ffva1156-2-e
                        NUM_OF_ACTIONS  = 1
                        HLS_SUPPORT     = FALSE
                        BRAM_USED       = TRUE
                        SDRAM_USED      = FALSE
                        NVME_USED       = TRUE
                        ILA_DEBUG       = FALSE
                        SIMULATOR       = xsim
                        USE_PRFLOW      = FALSE
                        ==============================
[PREPARE PROJECT.....] start 16:05:28 Fri Jul 06 2018
[PREPARE PROJECT.....] done  16:05:28 Fri Jul 06 2018
[SNAP PREPROCESS.....] start 16:05:28 Fri Jul 06 2018
                        configuring snap_core_types.vhd
[SNAP PREPROCESS.....] done  16:05:28 Fri Jul 06 2018
[CREATE PROJECT......] start 16:05:28 Fri Jul 06 2018
                        using Vivado v2017.4.1 (64-bit)
[CREATE_FRAMEWORK....] start 16:05:40 Fri Jul 06 2018
                        setting up project settings
                        importing design files
                        importing IPs
                        adding NVMe block design
                        adding NVMe Verilog simulation files
                        importing PSL design checkpoint
                        importing XDCs
[CREATE_FRAMEWORK....] done  16:06:17 Fri Jul 06 2018
[CREATE PROJECT......] done  16:06:18 Fri Jul 06 2018
[HW PROJECT..........] done  16:06:18 Fri Jul 06 2018
[COMPILE PSLSE ......] start 16:06:18 Fri Jul 06 2018
[COMPILE PSLSE ......] done  16:06:18 Fri Jul 06 2018
[COMPILE SOFTWARE....] start 16:06:18 Fri Jul 06 2018
[COMPILE SOFTWARE....] done  16:06:18 Fri Jul 06 2018
[... 93 lines cut ...]
                        using Vivado v2017.4.1 (64-bit)
[CREATE_FRAMEWORK....] start 16:07:40 Fri Jul 06 2018
                        setting up project settings
                        importing design files
                        importing IPs
                        adding NVMe block design
                        adding NVMe Verilog simulation files
                        importing PSL design checkpoint
                        importing XDCs
[CREATE_FRAMEWORK....] done  16:08:20 Fri Jul 06 2018
[CREATE PROJECT......] done  16:08:20 Fri Jul 06 2018
[HW PROJECT..........] done  16:08:20 Fri Jul 06 2018
[COMPILE PSLSE ......] start 16:08:20 Fri Jul 06 2018
[COMPILE PSLSE ......] done  16:08:20 Fri Jul 06 2018
[COMPILE SOFTWARE....] start 16:08:20 Fri Jul 06 2018
[COMPILE SOFTWARE....] done  16:08:20 Fri Jul 06 2018
[COMPILE APPLICATION.] start 16:08:20 Fri Jul 06 2018
[COMPILE APPLICATION.] done  16:08:20 Fri Jul 06 2018
[BUILD xsim MODEL....] start 16:08:20 Fri Jul 06 2018
                        patching SNAP version and build date registers
                        export simulation for version=2017.4.1
                        build a xsim model
[BUILD xsim MODEL....] done  16:09:45 Fri Jul 06 2018
bash-4.1$ 
```

fhaverkamp, 2018-07-06:
I changed nothing.

ThomasFuchs, 2018-07-09:
I tried to reproduce the failure, but without success. It is hard to debug it without log files! Please reopen the issue, if you see the problem again.

closed 2018-07-09 by ThomasFuchs

boekholt, 2018-07-09:
I also was not able to reproduce the problem. (I built a model, then modified  `hardware/sim/nvme_lite/nvme_top_i.sv` and rebuilt successfully)
@fhaverkamp Maybe your problem was caused by a previous simulation run that was not closed correctly/completely, which lead to problems with deletion of files?