open-power/snap #772: Vivado 2018.2: NVMe module synthesis fails

rs22, 2018-07-11:
When I try to build an image for the hls_nvme_memcopy action using Vivado 2018.2, the synthesis step fails. This is an excerpt from the logs:

```
INFO: [Synth 8-256] done synthesizing module 'nvme_top_util_buf_gte_1_0' (516#1) [~/snap/hardware/ip/nvme/nvme.srcs/sources_1/bd/nvme_top/ip/nvme_top_util_buf_gte_1_0/synth/nvme_top_util_buf_gte_1_0.vhd:65]
ERROR: [Synth 8-549] port width mismatch for port 'DDR_M_AXI_araddr': port width = 34, actual width = 32 [~/snap/hardware/hdl/core/psl_accel.vhd:687]
ERROR: [Synth 8-549] port width mismatch for port 'DDR_M_AXI_awaddr': port width = 34, actual width = 32 [~/snap/hardware/hdl/core/psl_accel.vhd:699]
ERROR: [Synth 8-285] failed synthesizing module 'psl_accel' [~/snap/hardware/hdl/core/psl_accel.vhd:203]
ERROR: [Synth 8-285] failed synthesizing module 'psl_fpga' [~/snap/hardware/hdl/core/psl_fpga.vhd:145]
```

I'm using this SNAP version:
```
$ git describe
v1.5.0-29-g1dc2b3f
```

boekholt, 2018-07-11:
With Vivado 2018.1 we do not run into this problem (here the `DDR_M_AXI_araddr` and `DDR_M_AXI_awaddr` port widths of the nvme bd match the port widths of the wrapper - all 32 bits) . We did not yet install Vivado 2018.2. This is one of our next work items. Will then look into why the widths do no longer match when using that Vivado version.

fhaverkamp, 2018-07-25:
I tried something like:
https://github.com/open-power/snap/commit/57dafacef82d1b856856cf30461a32aeb58b7f3a
when trying to circumvent the xsim problems I have with the NVMe_lite code.
I suppose I need other changes by Sven and Thomas.

boekholt, 2018-07-31:
The current `master` contains a circumvention for this.
@rs22 Please verify and close this issue.
Thanks

rs22, 2018-08-01:
Thanks, it also works for me!

closed 2018-08-01 by rs22