open-power/snap #785: 10140000_kill_test.sh timeout + unable to attach action

bmesnet, 2018-07-20:
10140000_kill_test.sh timeout on P8+8K5 and P9+RCXVUP and P9+NSA241 (Lu Yong's test)
```
[main] Exit rc:0
(1) Check if card is replying to a single memcopy request (1)..
action_wait_idle Timeout Error
err: Single snap_example took to long, please review results!
```

extending timeout with -t5 option get rids of timeout, but get to next error:
```
(4) Check if card is replying to a single memcopy request...
Error: Can not attach Action; 10140000
```

boekholt, 2018-07-20:
@bmesnet That is very difficult to understand (for me).
What is the exact sequence of commands you are executing?
Could you please paste the output of
```
software/tools/snap_maint -vvv
```

bmesnet, 2018-07-22:
I don't have the ability to do a screenshot on a P9 but here is the one I have on Marrennes (P8) with an ADK5). I don't have the timeout issue on this one. Could it be different on P9 due to the IRQ?
Did I missed a setting ? Thanks for taking care of that
 [image]

bmesnet, 2019-07-11:
Closing this issue since never reproduced again

closed 2019-07-11 by bmesnet