open-power/snap #797: SNAP config is corrupted by 'make image'

david-mle, 2018-08-14:
Hello,
recently my .snap_config file is getting overwritten with the contents of .snap_config.sh everytime a bitfile build is executed.
The code that triggers this behavior is line 30 of the NVMe Lite Makefile (the dependency of snap_config.sv).
Its not the code of that makefile target, its the dependency itself as far as I can tell. I don't know why this triggers the mentioned behavior, however, otherwise I would have tried to fix it...
https://github.com/open-power/snap/blob/7b6306ad21c9155c9f956fd1bf7ac5bbc582018c/hardware/sim/nvme_lite/Makefile#L30

My test procedure is the following sequence:
make menuconfig   => select a non-default card like ADKU3
make menuconfig   => check that previous config is unchanged as expected
diff .snap_config .snap_config.sh   => there is a difference as expected
make image
diff .snap_config .snap_config.sh   => there is no difference!
make menuconfig   => config is reset to default

Changing the above mentioned NVMe Lite dependency resolves the problem.

Best regards, David

boekholt, 2018-08-28:
Hi David,
Sorry for the delay caused by vacation time.
Unfortunately I am not able to reproduce your problem.
After following your instructions

> make menuconfig => select a non-default card like ADKU3
> make menuconfig => check that previous config is unchanged as expected
> diff .snap_config .snap_config.sh => there is a difference as expected
> make image

I do not see that `.snap_config` got overwritten.

Could you please provide more information regarding the configuration you selected?
And, could you please add some details regarding the console messages you receive?

Another question (just out of curiosity): Why do you use `make menuconfig` instead of `make snap_config`?

Thanks
Sven

david-mle, 2018-08-30:
Hi @boekholt,
No problem and sorry for my delayed reaction as well :)
I'll not be able to try before Monday, but will provide you a full session log. My checkout looked clean but maybe there were some files ignored by git. I'll use a fresh checkout.

The use of menuconfig was unintentional, I'm just used to it building Linux Kernels. I'll try to change to snap_config! I hope its not what is causing the problem, but the config gets altered by the following make image and I guess the config method used does not matter. We'll see.

Best regards, David

boekholt, 2018-09-03:
Hi @david-mle,

if `make menuconfig` would cause that problem then this is still something that has to be fixed.
But, I did exactly follow your description - also calling `make menuconfig` instead of `make snap_config` - and was not able to reproduce the problem. So, that is not the reason for the problem.

Calling `make snap_config` instead of `make menuconfig` just adds some checking and some automatic setup of `snap_env.sh`. So, it is recommended to call `make snap_config` but it is not necessary as long as the necessary environment variables are set up in some way.

david-mle, 2018-09-03:
Hello Sven,

I just reproduced the problem with a clean github checkout and it persists for me.
Attached is my session log: [snap.make_config_test.txt](https://github.com/open-power/snap/files/2344963/snap.make_config_test.txt)

The build machine is a Ubuntu 14.04 with GNU Make 3.81.
After login I enter a Vivado 2018.1 environment and clone the SNAP github repository.
In make snap_config I select the KU3 board and make no other changes. Then exit and save.
The scripts complain about PSL paths so I fix snap_env.sh (diff in log).
I run make snap_config again, to check the environment, but make no changes, just exit and save. KU3 board was still selected. No complaints by the scripts any more.
As you can see .snap_config and .snap_config.sh differ in file size (and content) at this point.
No I run make image and abort during synthesis with ctrl-c (behavior does not change if I let it complete, the config gets overwritten early on).
File sizes and content of .snap_config and .snap_config.sh are now the same and are the same as .snap_config.sh used to be in the beginning. The timestamp of .snap_config is the only one that changed.

My investigations so far show that this is done by the Makefile dependency outlined in my initial post.
I hope this helps to reproduce the problem.

Best regards, David

boekholt, 2018-09-05:
@david-mle Though I am using the same GNU Make version I am still not able to recreate the problem.
Nevertheless, the dependency `($SNAP_CONFIG)` in https://github.com/open-power/snap/blob/7b6306ad21c9155c9f956fd1bf7ac5bbc582018c/hardware/sim/nvme_lite/Makefile#L30 does not really make sense to me.
In the branch [Issue_797](https://github.com/open-power/snap/tree/Issue_797) I removed the dependency.
I opened pull request #808 for this branch.

david-mle, 2018-09-06:
Hello @boekholt,

your commit fixes the problem for me. Thanks!

Should you, out of curiosity, want to find the root cause of the initial problem: Using an older git commit I was able to reproduce it on the IBM Supervessel CAPI instance Xilinx2017.4-SNAP (which is CentOS based, so not an Ubuntu or specific machine issue). I guess you have access to such an instance. If you get to figure it out, I am interested in the explanation. Otherwise I consider the problem solved and don't really need to know why it existed in the first place :P

Best regards, David

boekholt, 2018-09-06:
I'd also be curious about the root cause. But an investigation regarding that will have very low priority (so, I guess I won't find the time for doing that).

Thanks for making us aware of this issue!

Closing the issue.

closed 2018-09-06 by boekholt