open-power/snap #802: IP cores generated by Vivado HLS not automatically included in SNAP project

TomHogervorst, 2018-08-28:
When an operation that requires the use of an IP core, like floating point addition, is used in an hls action, Vivado HLS will generate a vhdl file as toplevel for this IP core, and a tcl script to generate the IP core. However, currently it seems to me like this generated IP core is not added to the SNAP project during 'make model' or 'make image', resulting in the IP core being treated as a black box with no behaviour.

TomHogervorst, 2018-08-29:
Some more information about this issue:
It seems that `@$(MAKE) -s .create_ip_done` in the $SNAP_ROOT/hardware/Makefile below `create_project:` is not actually running the create_ip.tcl script. And when I force it to run this script, it will set action_vhdl to: ‘[SNAP_ROOT]/actions/hls_memcopy/hw/hlsMemcopy_xcku060-ffva1156-2-e/memcopy/sim/vhdl\n[SNAP_ROOT]/actions/hls_memcopy/hw/hlsMemcopy_xcku060-ffva1156-2-e/memcopy/syn/vhdl\n[SNAP_ROOT]/actions/hls_memcopy/hw/hlsMemcopy_xcku060-ffva1156-2-e/memcopy/impl/vhdl\n[SNAP_ROOT]/actions/hls_memcopy/hw/vhdl' , which causes `[file exists $action_vhdl] == 1 ` to fail, which in turn causes it to not import any user IP files.

My current solution/work-around to solve this problem is adding the following code to the import user IP section of create_framework.tcl:
`set tcl_exists [exec find $action_dir/ -name *.tcl]`
`if { $tcl_exists != "" } {`
  `foreach tcl_file [glob -nocomplain -dir $action_dir *.tcl] {`
    `set tcl_file_name [exec basename $tcl_file]`
    `puts "                        sourcing $tcl_file_name"`
    `source $tcl_file >> $log_file`
  `}`
`}`
Is this additional code actually necessary, or am I missing something?

boekholt, 2018-08-29:
@TomHogervorst could it be that your workspace was not clean when you started? Particularly could it be that the file `hardware/.create_ip_done` still existed from a previous run. This is the only reason I can think of for `create_ip.tcl` not getting executed.
Our idea behind that is that you only have to create the IPs once because in most cases they won't change. If you are actually adding IP after you created the framework, then you'd have to call `make clean` before calling `make model` or `make image` again.

TomHogervorst, 2018-08-29:
@boekholt You are right that I had not cleaned the workspace after adding the IP cores, which was the reason create_ip.tcl was not executed. However, because that script adds multiple paths to its $action_vhdl variable, it still fails to import the ip cores from any of those paths.

boekholt, 2018-08-29:
@TomHogervorst Another question regarding your second comment ("some more information..."):

I can't see how the statement
```
set action_vhdl  $action_root/hw/vhdl
```
can set `$action_vhdl` to the string containing multiple paths as you described.
Could it be that your environment variable `$ACTION_ROOT` (which should be defined in `snap_env.sh`) isn't set up properly?

TomHogervorst, 2018-08-30:
Oh, in the create_ip.tcl script that I'm using, action vhdl is set with the statement:
`set action_vhdl  [exec find $::env(ACTION_ROOT) -name vhdl]`
I must've been using an old version of this script. Thank you for your help and time.

closed 2018-08-30 by TomHogervorst