open-power/snap #817: "make model" imports tcl files (for user IPs) only during clean build

preelobo, 2018-09-17:
Ran into this error 
if we do a "make clean", and then "make model" the tcl files for user IPs from $action_root/hw/vhdl are imported into ~snap/hardware/ip/managed_ip_project/managed_ip_project.srcs/sources_1/ip using the "create_ip.tcl" script,
however, later on, if one updates their source code, resulting in new tcl files, and then doeas a "make model" (without make clean), the new tcl files are not imported, since there is the ".create_ip_done" file present already in ~snap/hardware did, which prevents the call for "create_ip.tcl".

This results into a scenario where $action_root/hw/vhdl has one set of tcl files/user ips
and ~snap/hardware/ip/managed_ip_project/managed_ip_project.srcs/sources_1/ip has another set resulting into "U"ndefined signals in simulation and black-box errors during opt_design step of FPGA image build.

Ideally, regardless of clean build or just plain build the latest tcl files from $action_root/hw/vhdl has to be imported

bmesnet, 2018-09-18:
This is mainly for HLS actions since HLS calls ip from libraries that can be different at every compilation.

boekholt, 2018-09-24:
We did decide that after adding or modifying SNAP IP you **always** have to create the Vivado project **from scratch**. I.e. you **need** to call `make clean` before calling `make hw_project` or `make model` or `make image`.
Support for HLS IP (user IP) got added recently. It got integrated into the SNAP IP generation process. This probably wasn't a good idea.  In branch `RecreateUserIp` I did change this approach. Please have a look into pull request #819.

boekholt, 2018-09-26:
I did decide to consolidate the approach of creation and addition of HDL actions and HLS actions. Therefore, I did close the pull request #819 without merging. I'll create a new branch and a new pull request.

boekholt, 2018-09-28:
After rebasing and modifying the support for HDL actions,  I created pull request #822 for this.

bmesnet, 2018-09-28:
Great thanks Sven. It perfectly answers the issue. Great work

bmesnet, 2018-11-06:
Preetham agreed problem was solved

closed 2018-11-06 by bmesnet