open-power/snap #866: HLS warning check: helloworld.log doesn't contain "setting up clock" message

luyong6, 2019-02-08:
In recent HLS checking improvements, 
in hls.mk, it goes to $(SOLUTION_DIR)*/$(SOLUTION_NAME)/$(SOLUTION_NAME).log  to look for something like: `Setting up clock 'default' with a period of 4ns.` but this log file doesn't contain this line. Instead, this line can be found in `vivado_hls.log`.

So the checking doesn't happen actually, and **/bin/sh: line 0: [: 4: unary operator expected** will be shown.

```
	@if [ $(HLS_ACTION_CLOCK) != $(shell grep "Setting up clock" $(SOLUTION_DIR)*/$(SOLUTION_NAME)/$(SOLUTION_NAME).log |cut -d " " -f 12|cut -d "n" -f 1) ]; then \
		echo " ---------------------------------------------------------- "; \
		echo " ERROR: Action was last compiled with a different HLS clock."; \
		echo " Please force the recompilation with a 'make clean' command";  \
		echo " ---------------------------------------------------------- "; exit -1; \
	fi
```

=========
hdclf037: luyong /afs/bb/u/luyong/capi/vol2/temp/master/actions/hls_helloworld/hw $ make
/afs/vlsilab.boeblingen.ibm.com/u/luyong/capi/vol2/temp/master/actions/scripts/create_run_hls_script.sh	\
		-n helloworld		\
		-d hlsUpperCase 		\
		-w hls_action			\
		-p xcvu3p-ffvc1517-2-i		\
		-c 4		\
		-f "action_uppercase.cpp " 			\
		-s /afs/vlsilab.boeblingen.ibm.com/u/luyong/capi/vol2/temp/master 		\
		-x """" > run_hls_script.tcl
vivado_hls -f run_hls_script.tcl >> /afs/vlsilab.boeblingen.ibm.com/u/luyong/capi/vol2/temp/master/hardware/logs/action_make.log
rm -f -rf hlsUpperCase_xcvu3p-ffvc1517-2-i/helloworld/syn/systemc hlsUpperCase_xcvu3p-ffvc1517-2-i/helloworld/syn/verilog
**/bin/sh: line 0: [: 4: unary operator expected**
   Checking for critical warnings during HLS synthesis .... OK
   Checking for critical timings during HLS synthesis  .... OK
   Checking for reserved MMIO area during HLS synthesis ... OK

bmesnet, 2019-02-11:
Thanks Lu Yong. I saw that once in the past but cannot have it recreated. I'll continue trying

bmesnet, 2019-02-15:
corrected by pull request #868

bmesnet, 2019-02-25:
add correction #872 since patch implemented in #868 works with HLS 2018.1 and not for 2018.2 and later.

bmesnet, 2019-05-02:
@luyong6 . this is an old issue corrected . tell me if we can close the issue

bmesnet, 2019-06-13:
Closing this issue since corrected

closed 2019-06-13 by bmesnet