open-power/snap #874: Valid flashdevice number on flash_mcs.tcl script for 9V3

diamantopoulos, 2019-03-26:
On trying to flash a factory image on a bricked 9V3, I've used flash_mcs.tcl script. However, I got an error for an invalid part number for the flash of 9V3. 

I was able to proceed by changing this
```
AD9V3 { set flashdevice mt25qu256aba8e12-1sit
        set fpgapartnum xcvu3p
         }
```
to
```
 AD9V3 { set flashdevice mt25qu256-spi-x1_x2_x4_x8
           set fpgapartnum xcvu3p
}
```
on https://github.com/open-power/snap/blob/4defea27569231f4e579c36bf3d0eac842081025/hardware/setup/flash_mcs.tcl#L80

Can you please verify if this change is valid? Thanks

diamantopoulos, 2019-03-26:
From 9V3 documentation: https://www.alpha-data.com/dcp/products.php?product=adm-pcie-9v3, on p.16 (ad-ug-1322_v2_7.pdf) we get:

[image]

... but later on p.17 we get:

[image]

, so it should be mt25qu256-spi-x1_x2_x4_x8 the correct device.

bmesnet, 2019-03-26:
You are correct when saying that you can hit this issue only when using the `flash_mcs.tcl`.  When using the hardware programmer GUI, you manually add the Flash device and don't use this `flash_mcs.tcl`. 

Thanks for finding that. Can you do the modification in `flash_mcs.tcl` and I will accept it?
I have checked that this change doesn't impact anything else in the whole SNAP code.

bmesnet, 2019-03-26:
fix in #875

closed 2019-03-26 by diamantopoulos