open-power/snap #891: Cannot simulate with QuestaSim

BasF0, 2019-07-17:
When trying to start simulation of an example with QuestaSim it fails with message:

> \# ** Error (suppressible): (vsim-19) Failed to access library 'work' at "work".
> \# No such file or directory. (errno = ENOENT)
> \# Error loading design

In the [compile_questa.log](https://github.com/open-power/snap/files/3401479/compile_questa.log) the first error  is:
> ** Error: ../../hdl/core/snap_core_types.vhd(257): near "address": (vcom-1576) expecting ':'.

Which fails to compile top, thus failing simulation. The line mentioned is a comment, so I'm not sure what is going on there.

occurs on latest master: 1507cb15cdce56b13f487cd3fc7562e24ba8d156
also tried one of the early commits mentioning questasim, same error: 1ebc208a0bc09a9eb9d1b41b80f7652e423b6d5a

Tried with FPGA= Xilinx U200, Nallatech 250S, hdl_helloworld, hls_helloworld examples..

Vivado v2018.3, QuestaSim-64 10.6a

bmesnet, 2019-07-22:
Hello
I have unfortunately no more access to this Questa license, but it seems that your issue could be due to the comment itself. Can you try and remove the [2 commented lines](https://github.com/open-power/snap/blob/master/hardware/hdl/core/snap_core_types.vhd_source#L266-L267), then do a `make clean` to be sure the file `snap_core_types.vhd `will be regenerated from the source file , and redo your simulation?
Thanks

BasF0, 2019-07-24:
Update: so removing the lines suggested only brings the error to the next "-- PSL *" comment until none of the source files in /snap/hardware/hdl/core/ have them anymore.. Found out why, looks like there is a naming conflict with https://en.wikipedia.org/wiki/Property_Specification_Language that Questa expects. To prevent this, in modelsim.ini I uncomment "EmbeddedPsl = 0"

Since "make model" resets the modelsim.ini file, atm I manually run vsim and run compile.do and elaborate.do

Also, the output is generated in a different directory than [run_sim seems to expect](https://github.com/open-power/snap/blob/master/hardware/sim/run_sim#L202-L203)
Whereas the output is at: 
```
$SNAP_ROOT/hardware/sim/questa/questa_lib/msim
$SNAP_ROOT/hardware/sim/questa/questa_lib/work
```

Also, I believe that simulation target should be "top_opt" and not "top" as in [run_sim](https://github.com/open-power/snap/blob/master/hardware/sim/run_sim#L143)

After this, calling run_sim, Questa seems to start without errors but pauses at 0ps. I believe that at this point a new terminal should pop up, but it doesn't. Manually hitting "run all" gives a fatal error on core/top.sv, "Null foreign function encountered when calling 'psl_bfm_init'. Maybe DCI related?

bmesnet, 2019-07-26:
Hi @BasF0 
Anything you can improve is welcome for the community. Don't hesitate to do a pull request of your code as soon as you have been able to fix this issue. I confirm that I can't get any license nor support for this simulator as of today. Maybe you Mentor Graphics can help you diagnosing the failure if you have a license/support. Thanks