open-power/snap #893: Action software.mk clean modification

shhssdm, 2019-07-23:
Hi CAPI-SNAP developers,

Would it be possible to modify the software.mk clean?

Currently, clean is only removing object files only in the same directory because of *.o argument.

However, my sw directory has several sub-directories inside and I'd like all object files inside them to be removed as well.

If would be nice if some statements in software.mk can be modified like *.o -> $($(projs)_objs).

Thanks in advance.

bmesnet, 2019-07-23:
Sure. Please propose some code doing a pull request and we will integrate it after regression tests. This is a community work and we are happy to get all contributions. 
Thanks for improving the code.

bmesnet, 2019-07-24:
Code is in pull request #894

bmesnet, 2019-07-24:
Code merged. Can we close this issue?
Thanks for improving the code

shhssdm, 2019-07-24:
Yes indeed. Thank you also.

closed 2019-07-24 by shhssdm