open-power/snap #898: There is no '/sys/class/cxl*'

lis123kr, 2019-09-09:
When I ran **'snap_find_card'** on Power 9, It didn't find my card.
I'm using **Power9 (AC922) and N250SP** card. So, I installed the FPGA card to Slot 3 P1-C3 of the server following the instruction (https://github.com/open-power/snap/blob/master/doc/README.md#p9-capi20-deployment-environment-bare-metal-ibm-server-examples-supporting-capi-snap-).
and I followed **QuickStart Guide** (https://github.com/open-power/snap/blob/master/doc/UG_CAPI_SNAP-QuickStart_on_a_General_Environment.pdf)
for setting the environment on Power 9 (Cloning the snap github and 'sudo apt-get install libcxl-dev')
However, I couldn't find '/dev/capi*' or '/sys/class/cxl*' on the server. But, I ran 'lspci | grep Nallatech' command, I can find it perceived. How can I find my card ?

**'lspci | grep Nallatech'**
> 0030:01:00.0 Serial controller: Nallatech Ltd. Device 250b

my server environments are below.

[OS]
> LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

[Firmware]

> Firmware version:
 Product Version       : witherspoon-ibm-OP9-v2.0-2.14-prod
 Product Extra         : bmc-firmware-version-2.01
 Product Extra         : buildroot-2018.02.1-6-ga8d1126
 Product Extra         : capp-ucode-p9-dd2-v4
 Product Extra         : hcode-b8173e8
 Product Extra         : hostboot-8ab6717d-p110cb65
 Product Extra         : hostboot-binaries-276bb70
 Product Extra         : linux-4.16.8-openpower2-p6a14c7f
 Product Extra         : machine-xml-7cd20a6
 Product Extra         : occ-77bb5e6
 Product Extra         : petitboot-v1.7.1-p50a5645
 Product Extra         : sbe-a596975
 Product Extra         : skiboot-v6.0.1

[Kernel Version]
> Linux 4.15.0-55-generic

bmesnet, 2019-09-09:
Hi @lis123kr
It seems that you did everything good on the software side but I guess you don't have a CAPI image in your FPGA card and so your card is not recognized by the system as a CAPI card. 
Reading again the explanations provided on our github, iI understand that you may have missed the point so I will arrange this shortly. I would suggest you to follow the [link how to program a blank card](https://github.com/open-power/snap/blob/master/hardware/doc/Bitstream_flashing.md#initial-programming-of-a-blank-or-bricked-card). Once the card is programmed and the server rebooted, you will get the followings:
> ~/snap$ **lspci | grep Nallatech**
~/snap$ **lspci | grep accel**
0006:00:00.0 Processing accelerators: IBM Device 0632 (rev 01)
0030:01:00.0 Processing accelerators: IBM Device 0477 (rev 02)
~/snap$ **snap_find_card -v -A N250SP**
N250SP card has been detected in card position: 0
 PSL Revision is                                                : 0x3001
 Device ID    is                                                : 0x0632
 Sub device   is                                                : 0x060d
 Image loaded is self defined as                                : user
 Next image to be loaded at next reset (load_image_on_perst) is : user
 Card PCI location is                                           : 0030:01:00.0
 Card PCI physical slot is                                      : WIO Slot1
~/snap$ **ls /sys/class/cxl**
afu0.0  afu0.0m  afu0.0s  card0
~/snap$ **ls /dev/cxl**
afu0.0m  afu0.0s

Let me know if this solve your issue.

lis123kr, 2019-09-10:
Thank you @bmesnet 

I think I need to buy 'Xilinx Platform programmer'(https://www.xilinx.com/products/boards-and-kits/hw-usb-ii-g.html) for programming a blank card. 

The instruction for programming a blank card said 
> The Nallatech N250S and N250SP require an additional Development & Debug Breakout Board 
to interface with the Xilinx Platform Cable USB II

Yes, I have another board with jtag, uart port and also USB port. I tried to program the flash using vivado tool.
>vivado -nolog -nojournal -mode batch -source setup/flash_mcs.tcl -tclargs "build/Images/${FPGACARD}_flash.mcs"

But, I've got 

>ERROR: [Labtoolstcl 44-199] No matching targets found on connected servers: localhost
Resolution: If needed connect the desired target to a server and use command refresh_hw_server. Then rerun the get_hw_targets command.
ERROR: [Common 17-39] 'get_hw_targets' failed due to earlier errors.
    while executing
"get_hw_targets"
    invoked from within
"llength [get_hw_targets]"
    invoked from within
"set numjtags [llength [get_hw_targets]]"
    (file "setup/flash_mcs.tcl" line 93)

When I googled it, I found out it needs 'Platform Cable USB II' and have to install 'cable driver'.
I'll let you know when I solve this problem with Platform Cable USB II.

bmesnet, 2019-09-10:
@lis123kr. 
From my knowledge, this is the only way to go for the 250S and 250SP. Some card vendors (like AlphaData) provide some drivers to rewrite the Flash without this programmer (but I never succeeded in using them) or even better has embedded the Xilinx programmer on the board itself (this work ok).
Let me know

bmesnet, 2019-10-01:
No news = good news? Please reopen this issue if the problem reaapears

closed 2019-10-01 by bmesnet

lis123kr, 2019-10-01:
@bmesnet 
I'm sorry, but we've been delayed due to the complicated purchasing process and delivery delay. I think I'll get it soon. 
Okay, If I have problem trying it, I'll reopen this issue.
Thank you