open-power/snap #900: ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

myheartisweeping, 2019-09-15:
An error happend when I run make image.  And I located at one sentence. It's because of float add and float multiply.  Here is the detail of error.
Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC INBB-3] Black Box Instances: Cell 'a0/action_w/hls_action_0/grp_process_action_fu_141/grp_CONVOLUTION_LAYER_SW_fu_1074/hls_action_fadd_3dEe_U2/hls_action_ap_fadd_6_full_dsp_32_u' of type 'hls_action_ap_fadd_6_full_dsp_32' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
ERROR: [DRC INBB-3] Black Box Instances: Cell 'a0/action_w/hls_action_0/grp_process_action_fu_141/grp_CONVOLUTION_LAYER_SW_fu_1074/hls_action_fmul_3eOg_U3/hls_action_ap_fmul_3_max_dsp_32_u' of type 'hls_action_ap_fmul_3_max_dsp_32' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
INFO: [Project 1-461] DRC finished with 2 Errors

bmesnet, 2019-09-16:
Hi Bruce
It seems that the IP you are calling is not found in Xilinx library. 
- Can you confirm that during the `make image` flow, you have the 3 lines I have highlighted below with `>> <<`?
- Are you using the latest master branch of the [snap git](https://github.com/open-power/snap)?
- Which Vivado release are you working with?
- Are you trying to build an image with  `make image `using the _HLS_decimal_mult_ action or are you using your own action code? If you could first try with the _HLS_decimal_mult_ action, this would validate the entire process first.

_(extract of the make image flow)_
```
[CREATE SNAP IPs.....] done  10:11:55 Mon Sep 16 2019                                                         
[CREATE HLS IPs......] start 10:12:04 Mon Sep 16 2019                                                         
                        >>sourcing hls_action_ap_fmul_3_max_dsp_32_ip.tcl<<                                      
                        >>generating HLS IP hls_action_ap_fmul_3_max_dsp_32<<                                     
[CREATE HLS IPs......] done  10:12:21 Mon Sep 16 2019                                                         
[CREATE_FRAMEWORK....] start 10:12:30 Mon Sep 16 2019                                                         
                        setting up project settings                                                           
                        importing design files                                                                
                        importing IP    
                        adding SNAP IP fifo_513x512                                                           
                        adding SNAP IP fifo_10x512                                                            
                        adding SNAP IP fifo_8x512                                                             
                        adding SNAP IP fifo_4x512                                                             
                        >>adding HLS Action IP hls_action_ap_fmul_3_max_dsp_32<<                                  
                        importing CAPI BSP (xcix)                                                             
                        importing XDCs                                                                                
```
FYI as a test to confirm the entire flow on this specific point, I have just rerun successfully the entire `make image `process with the _HLS_decimal_mult_ action and with the latest snap release using Vivado 2018.2.

myheartisweeping, 2019-09-16:
Thank you very much. @bmesnet. I use capi 1.0. So the snap is not the latest version. And my Vivado version is 2017.4.  I didn't use the _HLS_decimal_mult_ action but my own action code. I will update the software and try again. Thanks for your suggestions again.

bmesnet, 2019-09-16:
Thanks @myheartisweeping  for the answer. I have checked that our CAPI1.0 jenkins tests worked ok on this _HLS_decimal_mult_ action with Vivado 2018.1. Let me know how things works.

myheartisweeping, 2019-09-18:
@bmesnet. I changed the snap version. It worked. I used the latest snap version. And Vivado version is 2017.4.  Thank you very much.

bmesnet, 2019-09-18:
Great. Thanks for the feedback. closing the issue so. Have fun

closed 2019-09-18 by bmesnet