open-power/snap #906: Segfault when building simulation models for example actions

Deskarano, 2019-11-14:
I am trying to build some of the example actions for the Nallatech 250s. Specifically, I want to build the HLS Memcopy and the HLS NVMe Memcopy actions since these are most applicable to my use case. However, whenever I try to `make model` for these two actions I get a segfault when the xsim model is being built:

> [COMPILE APPLICATION.] start 14:35:42 Thu Nov 14 2019
[COMPILE APPLICATION.] done  14:35:43 Thu Nov 14 2019
[BUILD xsim MODEL....] start 14:35:43 Thu Nov 14 2019
                        patching SNAP version and build date registers
                        export simulation for version=2019.2
                        build xsim model
                        Error: please look into /home/grg/snap/hardware/logs/compile_xsim.log
Makefile:442: recipe for target 'xsim' failed
make[2]: *** [xsim] Error 255
Makefile:466: recipe for target 'model' failed
make[1]: *** [model] Error 2
Makefile:109: recipe for target 'model' failed
make: *** [model] Error 1

Here's my  [compile_xsim.log](https://github.com/open-power/snap/files/3847114/compile_xsim.log)

Through some debugging I've found that the faulting address is NULL, which indicates a simulator bug to me. I'm compiling on a machine with 60GB of RAM so I don't think this is an out-of-memory issue. This issue only happens for example actions that include DDR4 or NVMe. While this could be a Xilinx issue, I wanted to ask here if any of you had experienced this problem before.

Please let me know if I can provide any other information for debugging!

bmesnet, 2019-11-14:
Hello @Deskarano 
thanks for your interest in SNAP.
I just tested the hls_memcopy  and the hls_nvme_memcopy on Vivado 2019.1 on a N250S card (CAPI1.0 with DDR4 + NVME) with xsim simulator and don't hit this issue. 
Would you mind trying this with this release of Vivado. Indeed, I have no way to install the 2019.2 at this moment to be in the same condition than you.

Deskarano, 2019-11-15:
Hi @bmesnet 

Vivado 2019.1 worked flawlessly. I guess this is a bug with the newer version of xsim. Thank you!

bmesnet, 2019-11-15:
Thanks @Deskarano . 
BTW please pay attention that to build an image on Ultrascale FPGA family (CAPI1 cards) you may need to stay on an old 2018.1. Xilinx has disabled in next releases, the first Gen4 IP we were using for 250S card for example.
Refer to [vivado release ](https://github.com/open-power/snap/tree/master/doc#development-environments-) supported ok

bmesnet, 2019-11-28:
**Reopening the issue since it prevents from upgrading snap to Viavdo 2019.2 when DDR4 is needed**
Starting simulation data flow analysis
WARNING: [XSIM 43-4398] File "/snap/hardware/ip/ddr4sdram/rtl/cal/ddr4_v2_2_cal_cplx.sv" Line 855 : The System Verilog Assertion  will be ignored as it appears inside another process. This assertion is not yet supported.
WARNING: [XSIM 43-4398] File "/snap/hardware/ip/ddr4sdram/rtl/cal/ddr4_v2_2_cal_top.sv" Line 1783 : The System Verilog Assertion  will be ignored as it appears inside another process. This assertion is not yet supported.
Completed simulation data flow analysis
ERROR: [XSIM 43-3316] Signal SIGSEGV received.
Printing stacktrace...

[0] /Vivado/2019.2/bin/unwrapped/lnx64.o/xelab() [0x81bfb9]
[1] /xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/xelab() [0x829fb5]
[2] /xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/xelab() [0x46c584]
[3] /xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/xelab() [0x480965]
[4] /xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/xelab() [0x45241d]
[5] /usr/lib64/libc.so.6(__libc_start_main+0xf5) [0x2ad2680fe545]
[6] /xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/xelab() [0x46a3d0]

Looking for fix from Xilinx

bmesnet, 2021-03-16:
SR#10511625 opened 
Starting with 2019.2, I get an ERROR: [XSIM 43-3316] Signal SIGSEGV received when builiding the simulator model during the . Same with 2020.1.1 I have just tested

This issue is only related to DDR4. No issue with the same design with no DDR4
This issue is only related to xsim simulator. No issue when using a different simulator (xcelium for example)

DDR4 IP is created using the following line
create_ip -name ddr4 -vendor xilinx.com -library ip -version 2.* -module_name ddr4sdram -dir $ip_dir >> $log_file

closed 2021-05-04 by acastellane