open-power/snap #934: Generation of the IP CORE failed.

sysy007uuu, 2021-03-16:
When I run make model,it shows error:
source //home/sysy007uuu/Downloads/project_1/snap/actions/hdl_helloworld/ip/create_action_ip.tcl -notrace
[CREATE_ACTION_IPs..........] start 10:21:10 Tue Mar 16/ 2021
                        FPGACHIP = xcvu9p-flgb2104-2l-e
                        ACTION_ROOT = //home/sysy007uuu/Downloads/project_1/snap/actions/hdl_helloworld
                        Creating IP in //home/sysy007uuu/Downloads/project_1/snap/actions/hdl_helloworld/ip/action_ip_prj/action_ip_prj.srcs/sources_1/ip
                        Generating fifo_sync_32_512i512o ......
**ERROR: [IP_Flow 19-98] Generation of the IP CORE failed.
Failed to generate IP 'fifo_sync_32_512i512o'. Failed to generate 'Vivado VHDL Synthesis' outputs: 
INFO: [Common 17-206] Exiting Vivado at Tue Mar 16 10:21:15 2021...**
make[5]: *** [Makefile:26: config] Error 1
make[4]: *** [Makefile:24: hw] Error 1
                        Error: please look into //home/sysy007uuu/Downloads/project_1/snap/hardware/logs/action_make.log
make[3]: *** [Makefile:244: action_hw] Error 255
make[2]: *** [Makefile:316: .hw_project_done] Error 2
make[1]: *** [Makefile:492: model] Error 2
make: *** [Makefile:109: model] Error 1
while it shows like this?
My snap_env.sh is:
export PSLVER=9
export TIMING_LABLIMIT="-200"
export ACTION_ROOT=${SNAP_ROOT}/actions/hdl_helloworld
#export PSL_DCP= <for CAPI1.0 cards: path to b_route_design.dcp file>
export PSLSE_ROOT=//home/sysy007uuu/Downloads/project_1/pslse
export SNAP_ROOT=//home/sysy007uuu/Downloads/project_1/snap
export FPGACHIP=xcvu9p-flgb2104-2l-e
export CAPI20="y"
#export PSL9_IP_CORE= <for CAPI2.0 cards: path to ibm.com_CAPI_PSL9_WRAP_2.00.zip file>
export SIMULATOR=xsim

bmesnet, 2021-03-16:
Can you tell me  more ? Which vivado version are you using? which card are you targeting ?

sysy007uuu, 2021-03-17:
My vivado virsion is 2019.2, and my card is Alpha-Data ADM-PCIE-9V3 which support CAPI2.0

sysy007uuu, 2021-03-17:
I just configure the environment as Readme.md，but I can't run any example program in the snap/actions

acastellane, 2021-03-17:
Hi,
when simulating only, your self-made snap_env.sh file content should be limited to PSLSE_ROOT definition as the rest is set by the "make snap_config" step.
Check 
- https://github.com/open-power/snap/tree/master/hardware/sim#pslse-setup
- see page 4 of the Start Guide at : https://github.com/open-power/snap/blob/master/doc/UG_CAPI_SNAP-

Do you use the default xsim simulator ?
Thanks

sysy007uuu, 2021-03-17:
I had set it,but the still have the error

acastellane, 2021-03-17:
Hi,
using prepared examples, you should not mention manually the part reference (like <strike>export FPGACHIP=xcvu9p-flgb2104-2l-e</strike>). And this part is not the part that is used in AD9V3 card. So you might have some weird configuration.
You should restart from the beginning following the previously mentioned User's Guide.
Please use default "xsim" simulator for first tests.

bmesnet, 2021-03-17:
Your issue seems to be a Vivado installation issue. Let's check that things are correctly set on our side.
Let's start things as Alexandre suggested them from scratch
Can you try the following:
check that the 2 XILINX variables are correctly set => 
```
$ env |grep XILINX
	XILINX_VIVADO=<path>/xilinx/Vivado/2019.2
	XILINX_ROOT=<path>/xilinx
```
=> then install snap and pslse softwares and configure as follow
```
git clone https://github.com/open-power/snap.git
git clone https://github.com/ibm-capi/pslse.git
cd snap
make snap_config
```
=> select the card type **CAPI2 AlphaData AD9V3** and the action type **HLS HelloWorld** and keep all other options unchanged, then Exit
You should have the following displayed
```
=====Content of snap_env.sh============================
export TIMING_LABLIMIT="-200"
export ACTION_ROOT=${SNAP_ROOT}/actions/hls_helloworld
export PSLVER=9
#export PSL9_IP_CORE= <for CAPI2.0 cards: path to ibm.com_CAPI_PSL9_WRAP_2.00.zip file>
export PSLSE_ROOT=
=======================================================
```
Edit the snap_env.sh file to add pslse path as follo if you have put it at the same level than snap directory
`export PSLSE_ROOT=${SNAP_ROOT}/../pslse`

Then just build the simulation model and run the simulation executing the following:
```
make model
make sim 
```
Tell us if this doesn't work ok by sending us at what step it fails

sysy007uuu, 2021-03-18:
I change ubuntu to 20.04 to 18.04 than it works....

sysy007uuu, 2021-03-18:
hi, I reset my ubuntu system and do this again, now I do it as you say, my **snap_env.sh** is(**I now change it from hld_helloworld to hls_helloworld**):
_export PSLVER=9
export TIMING_LABLIMIT="-200"
export ACTION_ROOT=${SNAP_ROOT}/actions/hls_helloworld
export PSLSE_ROOT=${SNAP_ROOT}/../pslse
export XILINXD_LICENSE_FILE=/root/.Xilinx/vivado.lic
export CAPI20="y"
#export PSL9_IP_CORE= <for CAPI2.0 cards: path to ibm.com_CAPI_PSL9_WRAP_2.00.zip file>
export SIMULATOR=xsim_
when I run make model, the error is:
_[HW PROJECT..........] start 17:19:55 Thu Mar 18 2021
[CONFIG ACTION HW....] start 17:19:55 Thu Mar 18 2021
   Compiling action with Vivado HLS v2020.1
   Clock period used for HLS is 4 ns
/home/sysy007uuu/Downloads/fpga/snap/actions/hls.mk:67: recipe for target 'hlsUpperCase_/helloworld/syn' failed
make[5]: *** [hlsUpperCase_/helloworld/syn] Error 1
Makefile:24: recipe for target 'hw' failed
make[4]: *** [hw] Error 1
                        Error: please look into /home/sysy007uuu/Downloads/fpga/snap/hardware/logs/action_make.log
Makefile:241: recipe for target 'action_hw' failed
make[3]: *** [action_hw] Error 255
Makefile:316: recipe for target '.hw_project_done' failed
make[2]: *** [.hw_project_done] Error 2
Makefile:492: recipe for target 'model' failed
make[1]: *** [model] Error 2
Makefile:109: recipe for target 'model' failed
make: *** [model] Error 1_

sysy007uuu, 2021-03-18:
Can you tell me how to choose I card?  Should I write it in snap_env.sh? I can't run **hdl_helloworld** because it shows:

_root@ubuntu:/home/sysy007uuu/Downloads/fpga/snap# make model
=======================================================
== Precompiling the Action logic: hdl_helloworld
=======================================================
[HW PROJECT..........] start 17:25:23 Thu Mar 18 2021
[CONFIG ACTION HW....] start 17:25:23 Thu Mar 18 2021
Call action_config.sh (creating action IPs)
                        action config says ACTION_ROOT is /home/sysy007uuu/Downloads/fpga/snap/actions/hdl_helloworld
                        action config says FPGACHIP is 
                        Call create_action_ip.tcl to generate IPs

****** Vivado v2020.1 (64-bit)
  **** SW Build 2902540 on Wed May 27 19:54:35 MDT 2020
  **** IP Build 2902112 on Wed May 27 22:43:36 MDT 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source /home/sysy007uuu/Downloads/fpga/snap/actions/hdl_helloworld/ip/create_action_ip.tcl -notrace
[CREATE_ACTION_IPs..........] start 17:25:31 Thu Mar 18/ 2021
                        FPGACHIP = 
                        ACTION_ROOT = /home/sysy007uuu/Downloads/fpga/snap/actions/hdl_helloworld
                        Creating IP in /home/sysy007uuu/Downloads/fpga/snap/actions/hdl_helloworld/ip/action_ip_prj/action_ip_prj.srcs/sources_1/ip
ERROR: [Coretcl 2-106] Specified part could not be found.
INFO: [Common 17-206] Exiting Vivado at Thu Mar 18 17:25:31 2021...
Makefile:25: recipe for target 'config' failed
make[5]: *** [config] Error 1
Makefile:24: recipe for target 'hw' failed
make[4]: *** [hw] Error 1
                        Error: please look into /home/sysy007uuu/Downloads/fpga/snap/hardware/logs/action_make.log
Makefile:241: recipe for target 'action_hw' failed
make[3]: *** [action_hw] Error 255
Makefile:316: recipe for target '.hw_project_done' failed
make[2]: *** [.hw_project_done] Error 2
Makefile:492: recipe for target 'model' failed
make[1]: *** [model] Error 2
Makefile:109: recipe for target 'model' failed
make: *** [model] Error 1_

sysy007uuu, 2021-03-18:
It just means I should select a **FPGACHIP** manually

sysy007uuu, 2021-03-18:
I put the Power9 IP core in the default log ${SNAP_ROOT}/hardware/capi2-bsp/psl/ibm.com_CAPI_PSL9_WRAP_2.00.zip, so I don't need to set it manually.

sysy007uuu, 2021-03-18:
I update my Vivado from 2019.2 to 2020.1

acastellane, 2021-03-18:
Hi,
to simulate from scratch you need to run _make snap_config_ as per doc or Bruno's message. This will allow you to select the card. Please use 2019.2 version of Vivado.
Best regards

sysy007uuu, 2021-03-18:
what does **scratch you need to run make snap_config as per doc or Bruno's message** mean?

acastellane, 2021-03-18:
that means, restart with 2019.2 from clean git installed default directories and following the steps mentioned in the User's Guide, or Bruno's very quick steps. I noticed that the hardware/sim/README.md doesn't mention the _make snap_config_ step, I'll correct this.

sysy007uuu, 2021-03-18:
So I shouldn't edit snap_env.sh manually？

bmesnet, 2021-03-18:
Edit the snap_env.sh file to add **pslse root path** as follow (if you have put it at the same level than snap directory)
`export PSLSE_ROOT=${SNAP_ROOT}/../pslse`
You may need to add also the path to your zip file, but this is needed only for synthesis (not useful for simulation)
`export PSL9_IP_CORE= <for CAPI2.0 cards: path to ibm.com_CAPI_PSL9_WRAP_2.00.zip file>`

bmesnet, 2021-04-30:
Hello @sysy007uuu 
Is this question answered? Have you been able to have it work?
If yes, please let close the question
thx

closed 2021-05-04 by acastellane